tronfu/Voron-Mods #1: docklocation_z = ?

bjoern79de, 2022-03-01:
It would be nice, if docklocation_z could be determined automaticly (by using the actual probe-attachment as a second z-endstop). One of the main advantages of klicky-probe is to use it together with klipper_z_calibration, to be able to replace nozzles/printheads without any reconfigurations. I have height-differences of up to 10mm between my printheads - so I would have to reconfigure the docklocation_z instead of z-offset every time.

tronfu, 2022-03-01:
Thanks for the suggestion. Is this a problem only with the servo-dock or also with the stock gantry-mounted dock?

bjoern79de, 2022-03-02:
Only with a non-gantry-mounted dock. With gantry-mounted docks, the z-location is not relevant.

tronfu, 2022-03-02:
Understood. When you say using the actual probe-attachment as a second z-endstop, what exactly are you envisioning? Can you visually illustrate that for me?

tronfu, 2022-03-03:
This functionality has been added. See the latest README.